WOA-Project/MSM8994-8992-NT-ARM64-Drivers #8: (GPU driver）Direct x11  0x887A0006 - DXGI_ERROR_DEVICE_HUNG

yifengling0, 2019-03-20:
Lumia950XL with GPU dirver working。
OS : 17134

I wrote an  desktop app(arm/arm64） with Direct X11.

I got 0x887A0006 - DXGI_ERROR_DEVICE_HUNG error.

I also checked the dx support with dxdiag command.  no error.

Is this a known issue?

woachk, 2019-03-20:
Hi, which application?

yifengling0, 2019-03-20:
https://github.com/yifengling0/angle

with this one. i had port to arm & arm64. may be i made some mistake.

gus33000, 2019-03-23:
Hi, x86 emulation has an issue because we do not have user mode components for x86. It's not related to this issue. I'll investigate the issue with your app hopefully sometime this weekend. Sounds weird it runs well on a Lumia 930, since the Lumia 930 supports DirectX 9 on Windows RT (by design). Are you sure you've built the app for Direct X 11?

yifengling0, 2019-03-24:
> Hi, x86 emulation has an issue because we do not have user mode components for x86. It's not related to this issue. I'll investigate the issue with your app hopefully sometime this weekend. Sounds [...]

thanks for reply .  about x86 emulation cann't use 835 user mode  components?  
directx 11 is working on rt .  i am sure i am using the directx11 api.

woachk, 2019-03-24:
@yifengling0 835's DX shader compiler (and even 850) component is reusable, the DDI isn't because support for Adreno 4xx was removed.

yifengling0, 2019-03-27:
> @yifengling0 835's DX shader compiler (and even 850) component is reusable, the DDI isn't because support for Adreno 4xx was removed.

thanks for replay。 since UWP can working well。 can we rewrite the ddi for Adreno4XX？

gus33000, 2019-04-07:
Turns out we're not properly registering the directx 11 user mode components properly, fixing that hopefully today.

yifengling0, 2019-04-20:
> Turns out we're not properly registering the directx 11 user mode components properly, fixing that hopefully today.

good new，is there any progress？

gus33000, 2019-05-03:
> > Turns out we're not properly registering the directx 11 user mode components properly, fixing that hopefully today.
> 
> good news， any progress？

Turns out the issue is slightly complex, we got the user mode direct x compiler crashing every minute or so which is also the root cause of the black box glitches. Will need to be looked into, but registration of components was also originally a problem

woachk, 2019-05-17:
Fixing would involve switching to a newer DX UMD version, which is harder than it sounds. Hopefully wouldn't have issues with 64-bit pointers when using dxbridge that is being developed for x86 compatibility anyway...

woachk, 2019-05-26:
Can you test an ARM32 version with the 1906-directx branch?

Thanks,

yifengling0, 2019-05-27:
> Can you test an ARM32 version with the 1906-directx branch?
> 
> Thanks,

OK, i will try later.

yifengling0, 2019-10-27:
> Can you test an ARM32 version with the 1906-directx branch?
> 
> Thanks,

I test a desktop arm32 application with newer version . but also got the same error!

yifengling0, 2019-10-29:
> > Can you test an ARM32 version with the 1906-directx branch?
> > Thanks,
> 
> I test a desktop arm32 application with newer version . but also got the same error!

but with the RDP the application can run normally.