WOA-Project/MSM8994-8992-NT-ARM64-Drivers #10: Audio and other drivers for Hapanero pre-EB2

PC-DOS, 2019-04-21:
In recent tests, we've found that  some drivers provided for Hapanero EB2 and/or 950XL, including audio drivers, can be directly applied to Hapanero pre-EB2.

However, the SD card may still don't work, inserting a card has no reactions. I'm not sure if it's caused by qcPEP or something else I don't have a WP image which was extraced from pre-EB2 Hapaneroes with a SD card support (parameters for SDC seems to be missing from the registry). BSOD with SDBUS_INTERNAL_ERROR is also a common problem.

These tests were done with 1902 drives, on Win10 17134. Drivers used for testing can be found at [This Repo](https://github.com/PC-DOS/RX-130-EB1.0-Win10-ARM64/tree/testing). Using drivers provided with 1903 or 1904(main branch) may run into some problems including BSOD, weird Wi-Fi or GPU behavior, etc. We are still testing, but finding a resolution may be hard for us.

Thanks a lot!

CornyjK, 2019-04-21:
I don't know much about those drivers, my answer might not be accurate/relevant, but IIRC drivers are same for all devices (exceptions apply), just some of them may have tweaks for specific device.

woachk, 2019-04-21:
@PC-DOS for Hapanero EB2, problems are present with drivers that depend on qcsubsys. Other drivers that do not depend on subsys can work, will actualize that info later too.

gus33000, 2019-04-26:
@PC-DOS It's normal for you to get issues with EB2 drivers as QcSubsys won't work nicely on Cobalt 1 SoCs (such as MSM8994AA) I agree there's room for improvements regarding EB2, but due to a lack of a testing device I can't make much progress on that. I'll probably see with @woachk when he's free about that (and when I'm free too). Looked quickly at your repo, some drivers you install are either not needed or won't work as is. We'll clarify more later (when we will have time). For now keep in mind audio requires subsys without it it won't work. I also saw you tried to install some inbox windows drivers, I don't understand at all why..

PC-DOS, 2019-04-30:
@gus33000 The driver tree in my repo was created by an all-in-one tool (Dism++) and I haven't organized them (for faster testing). I apologize for that.

The inbox drivers were used for some early tests and they are not included in recent tests. As in early tests, the Windows (with no third party drivers deployed) will crash before OOBE. Please ignore these inbox drivers...

QcSubSys, as well as audio, was successfully launched on 17134 with drivers from the February version of this driver package. Tmsix (Mr. Chen) told me that he will (or already have) send the OS image to you along with an EB1 Hapanero for testing.

woachk, 2019-05-01:
@PC-DOS is your EB1 HWID1000 or a later revision?

PC-DOS, 2019-05-01:
@woachk Yes, its HWID is 1000

gus33000, 2019-05-10:
[image]
Issue is being dealt with ;)

gus33000, 2019-05-10:
Problem resolved, please use the 1905-pre branch for Hapanero EB1.
Inject  MSM8994 folder as well as Device Specifics\Hapanero\EB1 and Device Specifics\MMO to get feature parity with EB2 RX130. Everything works the same including cellular. To get GPU to work install under remote desktop the drivers provided under "QcomPkg - EB1 - Install From RDP"

commit cc82aaa7 (2019-05-10): Final commit for 8994 AA, and with this, we fix #10
closed 2019-05-10 by gus33000