WOA-Project/MSM8994-8992-NT-ARM64-Drivers #13: [950XL Single Sim] Sim Card Not Detected.

raidensnake, 2019-05-09:
The Mobile device status is showing "Account Not Active" When trying to use my EE Sim card.

CornyjK, 2019-05-09:
Celluar does not work yet.

CornyjK, 2019-05-09:
I thought this is deployer repo for some reason, reopening as I am not supposed to close issues here.

woachk, 2019-05-09:
@CornyjK issue will be closed soon anyway, cellular is up but that isn't pushed here yet.

CornyjK, 2019-05-09:
> 
> 
> @CornyjK issue will be closed soon anyway, cellular is up but that isn't pushed here yet.

I know, I have seen the great efforts and the result.

raidensnake, 2019-05-09:
If you guys manage to get the mobile side fully working and the cameras working. I'd throw out wp10 altogether!

gus33000, 2019-05-10:
GPS works already since a while on 950 XL, Hapanero devices.  Camera is on the todo list but we can't say for sure if they will ever work. Cellular has been brought up already but issues exist. As issues exists still even though sim cards are detected, I'm not going to close this issue until that's dealt with. See #11 for one of the issues.

raidensnake, 2019-05-10:
nice to hear. Any updates on the sound volume fix?

gus33000, 2019-05-17:
Sound is a bit tricky, so for now, no no updates on that

closed 2019-05-17 by gus33000