WOA-Project/MSM8994-8992-NT-ARM64-Drivers #16: The phone does not charge in Windows 10. Телефон не заряжается в Windwos10.

Hemp007, 2019-05-29:
The phone does not charge in Windows 10. What could be the reason?

Телефон не заряжается в Окна 10. В чем может быть причина?

woachk, 2019-05-31:
Does it charge with the screen off under Connected Standby? Which Woa Deployer version did you use?

Hemp007, 2019-06-01:
Yes, in the off state there is a charge, but then the phone turns on automatically and the OS does not charge. I used the assembled FFU image and installation via the command line.

feherneoh, 2019-06-01:
> Yes, in the off state there is a charge, but then the phone turns on automatically and the OS does not charge. I used the assembled FFU image and installation via the command line.

That wasn't the question.
It wasn't about the phone, but the display being turned off (so basically the phone locked)

Hemp007, 2019-06-01:
> > Yes, in the off state there is a charge, but then the phone turns on automatically and the OS does not charge. I used the assembled FFU image and installation via the command line.
> 
> That wasn't the question.
> It wasn't about the phone, but the display being turned off (so basically the phone locked)

No, with a locked screen, the charge does not go, there is no charge indicator.

feherneoh, 2019-06-01:
> No, with a locked screen, the charge does not go, there is no charge indicator.

if you turn the display off, you cannot see if there is an indicator or not

the indicator usually takes about 2 minutes to appear if it appeares at all, but the phone should still charge while the display is off, even if there is no indication of it

if it doesn't, then the problem is probably with the specific charger you are using
Lumia 950XL stock charger usually works, Xiaomi/Asus 2A chargers seem to work even while the screen is on (using the stock C cable)

Hemp007, 2019-06-02:
The cable is working, the charge is only for wireless charging.
The indicator meant in the tray, it is not from the cable, there is a wireless charging.
Yesterday I reinstalled via WOA Deployer, nothing has changed.

gus33000, 2019-06-15:
Possible consequence of #28

gus33000, 2019-08-28:
Issue fixed in 1909 (Should be released next week)

closed 2019-08-28 by gus33000