WOA-Project/MSM8994-8992-NT-ARM64-Drivers #22: GPU driver often creates black screen issues and display glitches on MSM8992

gus33000, 2019-06-15:


DremOSDeveloperTeam, 2019-08-09:
i have the same problem. i get black screen issues on >=18362, and i get strange square artifacts in anything that has a transparent effect, such as the action center.

CxZMoE, 2019-08-18:
> 
> 
> i have the same problem. i get black screen issues on >=18362, and i get strange square artifacts in anything that has a transparent effect, such as the action center.

So i just turn it off...

DremOSDeveloperTeam, 2019-08-18:
> > i have the same problem. i get black screen issues on >=18362, and i get strange square artifacts in anything that has a transparent effect, such as the action center.
> 
> So i just turn it off...

I mean black flickering. Rebooting doesn't fix that.