WOA-Project/MSM8994-8992-NT-ARM64-Drivers #23: WLAN is not working on MSM8992

gus33000, 2019-06-15:


quiret, 2019-06-25:
Mysterious tweet by gus33000 showing WLAN support, probably Talkman: https://twitter.com/gus33000/status/1143495664510361600

woachk, 2019-08-16:
Will be fixed tomorrow.

Arturbc, 2019-08-17:
Very nice! This works now!

Thanks!

CxZMoE, 2019-08-17:
Nice work!!!!

closed 2019-08-17 by woachk

levyst, 2019-08-24:
how to install plz {WLAN 950]

Arturbc, 2019-08-25:
> how to install plz {WLAN 950]

Download the lastest release preview on releases or get files from WLAN folder directly from git. After download complete, extract files, if you download as .zip, and use Device Manager to update your WLan driver.

levyst, 2019-08-25:
Thanks alot,now my main problem is how to connect I cant connect my phone to computer using  usb type c cable, may be to ask which usb mode  should I select to connect coz have tried both accept the second from top.

Have 3 more issues to ask.

1. battery it take time to charge and when full it cant last more than 20 minutes in usage.it drains first
2.space,out of 18gb after installation I only remains with 4gb of storage.

3.I can see  boxed like images while on use may be coz of Gpu.

All this happens to Talkman.

I do appreciate great work you are doing .

Thanks regards.

Levy

CornyjK, 2019-08-25:
Do not hijack threads, if you have any questions, use Telegram group!

levyst, 2019-08-26:
Noted.

Thanks