WOA-Project/MSM8994-8992-NT-ARM64-Drivers #42: Please, could you explain, how to install this driver package on a Lumia 950XL, which already has Windows 10 1903 ARM64.

pachnes, 2019-07-13:
Please, could you explain, how to install this driver package on a Lumia 950XL, which already has Windows 10 1903 ARM64.

woachk, 2019-07-13:
Driver pack build to build updates aren't currently supported.

woachk, 2019-07-13:
Issue renamed, remember that we don't even change driver versions on build updates, which will make this harder. Will have to talk on if we'd even attempt to do it or not.

pachnes, 2019-07-13:
I did a new install and watched what the WOA.Deployer did do. The WOA.Deployer does download and "inject" the drivers during installation . Could WOA.Deployer get modified, that there is a menu to "inject" a new driver paket in UEFI Mass storage mode"?

quiret, 2019-08-22:
I like the idea of @pachnes. It is pretty hard to especially update the GPU driver (the OS crashes during runtime).

woachk, 2019-08-23:
@pachnes @quiret I know that some drivers at least cannot be even replaced via injection after the fact. (one of those is PEP, which makes things... fun)