WOA-Project/MSM8994-8992-NT-ARM64-Drivers #51: Bluetooth

pachnes, 2019-09-06:
Hi Gus, you should advise the users, to use a cheap blueetooth keyboard with touchpad to use the device after initial configure. There are a few keyboards with touchpad available available at a price between 20,-- and 35,-- Euros. I do not recommend to use separate bluetooth keyboards and separate bluetooth mice, as bluetooth still is not very stable, A single bluetooth device is stable.

lwlsw, 2019-09-06:
说的很对，现在的蓝牙非常不稳定，经常性的掉线，我用蓝牙鼠标和蓝牙键盘，掉线的时候一起掉线。还希望能修复一下

woachk, 2019-09-06:
@lwlsw ok, I'll see if there'd be a way to make it more stable, but not too optimistic on that one.

lwlsw, 2019-09-06:
@woachk 恩，辛苦了，非常感谢您的辛苦付出。

SuperJMN, 2019-09-06:
@lwlsw Please, write in plain English. Most people won't understand what you say.

lwlsw, 2019-09-06:
@SuperJMN Ok, I will do in the future.

pachnes, 2019-09-06:
lslsw did write "That's right, the current Bluetooth is very unstable, and it is Occasionally disconnect. I use a Bluetooth mouse and a Bluetooth keyboard. They are Occasionally disconnect together, Also hope to fix it."

That is, why I recoomend a bluetooth keyboard device which has an integrated touchpad. If there is only 1 bluetooth connection, bluetooth is quiet stable.

pachnes, 2019-09-06:
woachk, or Gus, I did write this at "Issues". I know, that this is the wrong place to suggest something. Are you able to put a new Tab with the name "Discussion" here?

gus33000, 2019-09-08:
Issue might be related to Windows classic Bluetooth problems since RS5 release. Won't close just yet as I need to confirm if it's Windows OS related, if it is, I will close this issue and advise people to file a bug report in the Feedback Hub or upvote an existing one like this one: https://aka.ms/AA60c18
[image]

quiret, 2019-09-08:
@gus33000 MSFT might have changed Feedback Hub so that you cannot update "Problem"s anymore. Interesting.

falkor2k15, 2019-10-02:
Go back to Device manager.
Click on the Bluetooth drop down.
*Right-click on your Bluetooth Adapter choose Properties.
Go to the Power Management Tab.
Uncheck Allow the computer to turn off this device to save power.

*Note: it's called something like "XXX radio service" (the 2nd one in the list)

yifengling0, 2019-11-22:
> Go back to Device manager.
> Click on the Bluetooth drop down.
> *Right-click on your Bluetooth Adapter choose Properties.
> Go to the Power Management Tab.
> Uncheck Allow the computer to turn off this device to save power.
> 
> *Note: it's called something like "XXX radio service" (the 2nd one in the list)

thanks for your hit。 working but system went unstable( dead with block screen).
and I deleted the driver any more.  Bluetooth working well with no problem. WOW!!

gus33000, 2019-12-29:
Closing as the issue is windows related

closed 2019-12-29 by gus33000