WOA-Project/MSM8994-8992-NT-ARM64-Drivers #55: Attempting to Use Cellular, "Account Not Active"

DremOSDeveloperTeam, 2019-10-07:
I've seen some videos on YouTube where cellular is working, but my device still says "Account Not Active." Are the drivers being worked on but just haven't been pushed to the repository, or am I missing something? I'm using T-Mobile if that helps.

I know this issue is quite similar to [this one](https://github.com/WOA-Project/MSM8994-8992-NT-ARM64-Drivers/issues/11), but I see the IMEI number and other things in the settings for cellular, so I thought that I may have the drivers for cellular but there may be an issue with T-Mobile. Thanks!

gus33000, 2019-10-09:
Your issue is the same issue as #24 Closing as this is a duplicate. If you want cellular to work use build 18363 max.

closed 2019-10-09 by gus33000

DremOSDeveloperTeam, 2019-10-18:
oh, thanks, didnt see that 18363 is the maximum build that works with cellular in that issue so i decided to make another.