WOA-Project/MSM8994-8992-NT-ARM64-Drivers #68: Lumia 950 turn into blue screen first boot after deploy:driver pnp watchdog

CxZMoE, 2020-01-28:
Installed Image:19551.1005.200119-2325.RS_PRERELEASE_FLT_CLIENTPRO_OEMRET_A64FRE_ZH-CN.ISO
Machine Type: Lumia 950XL
Stop Code: driver pnp watchdog

CxZMoE, 2020-01-28:
I chose this from uups:Windows 10 Insider Preview 10.0.19551.1005 (rs_prerelease) arm64

CxZMoE, 2020-01-28:
and now I'm trying 19456

CxZMoE, 2020-01-28:
and it still the same thing. now try 18950

SuperJMN, 2020-01-28:
Don't try anything newer than 19041.

SuperJMN, 2020-01-28:
Important note: Manganese builds are unsupported, the maximum build you can install and use is build 19041

SuperJMN, 2020-01-28:
Not related to WOA Deployer

closed 2020-01-28 by SuperJMN

CxZMoE, 2020-01-29:
> Important note: Manganese builds are unsupported, the maximum build you can install and use is build 19041
> […](#)
> ________________________________ From: José Manuel Nieto Sánchez [...]

thanks a lot!!!